shaxxx/flutter_barcode_listener #1: The bar code is missing and cannot be identified completely

zouxianbincc, 2021-03-01:
Hello
I'm glad to find your open source project
However, during the testing process,
The bar code is missing and cannot be identified completely

shaxxx, 2021-03-01:
Not sure what exactly is the problem?
Is barcode missing or is it not identified completely? These are two distinct cases.
If you're missing part of the barcode have you tried to specify `bufferDuration ` parameter to something longer, IE. 500 ms?

zouxianbincc, 2021-03-01:
Bar code missing
Correct: 6901028225168
Identified: 690028225158
There are fewer digits in the middle

shaxxx, 2021-03-01:
Have you tried modifing bufferDuration parameter?

zouxianbincc, 2021-03-02:
const Duration aSecond = Duration(seconds: 1);
const Duration hundredMs = Duration(milliseconds: 500);
const int lineFeed = 10;

我改了hundredMs=500
但是测试结果，还是一样。
数字还是出现缺失

zouxianbincc, 2021-03-02:
Thank you very much.
After testing, change RawKeyUpEvent to RawKeyDownEvent
It can be correctly identified

zouxianbincc, 2021-03-02:
void _keyBoardCallback(RawKeyEvent keyEvent) {
    if (keyEvent.logicalKey.keyId > 255 &&
        keyEvent.data.logicalKey != LogicalKeyboardKey.enter) return;
    if (keyEvent is RawKeyDownEvent) {
      if (keyEvent.data != null) {
        if (keyEvent.data is RawKeyEventDataAndroid) {
          _controller.sink.add(((keyEvent.data) as RawKeyEventDataAndroid).codePoint);
        } else if (keyEvent.data is RawKeyEventDataFuchsia) {
          _controller.sink
              .add(((keyEvent.data) as RawKeyEventDataFuchsia).codePoint);
        } else if (keyEvent.data.logicalKey == LogicalKeyboardKey.enter) {
          _controller.sink.add(lineFeed);
        }
      } else {
        _controller.sink.add(keyEvent.logicalKey.keyId);
      }
    }
  }

shaxxx, 2021-03-04:
Glad it worked. If you check history you'll see I started with KeyDown but moved to KeyUp because there were some issues. 
Don't remember why anymore.

closed 2021-03-04 by shaxxx