shaxxx/flutter_barcode_listener #5: connect to external barcode scanner does not work

lvchaoin, 2021-08-21:
Hi there,

I'm using a android device which needs to connect an external barcode scanner, the scanner is an scanner box connect to the android PAD, just like the device in supermarket cashier machine computer.  tried a long time, but still not work, I wonder if there's special settings need to be done in the scanner side ?

Actually when I tried in emulator, and input characters & enter key in the computer keyboard rapidly, it seems work.

Thanks a lot !

shaxxx, 2021-08-21:
So it works?

lvchaoin, 2021-08-21:
no, I mean it works with physical key press in the computer, but not work with scanner connected to android real device, the scanner has enter key setting in the barcode ending. How can I proceed to troubleshoot ?

shaxxx, 2021-08-21:
Connect barcode scanner to android device
Open some text editor app
Try to scan barcode

If you dont see barcode appear as text in editor when you scan your barcode it means barcode scanner is not working properly

lvchaoin, 2021-08-21:
with text editor it's ok, I can see the barcode content, now the issue is I can't get the result with the example.

shaxxx, 2021-08-21:
Put a breakoount on this line
https://github.com/shaxxx/flutter_barcode_listener/blob/486d3b48ac77cde40e4e7cf4926165af8215ada0/lib/flutter_barcode_listener.dart#L70
And see if it gets hit when you scan

lvchaoin, 2021-08-23:
Thanks a lot. I get the problem solved, it seems due to the external scanner feature.
[image]

One another question, from troubleshooting, the barcode callback always return lower case string, how can I keep the original value(case sensitive) ? and I put a special character "@" in the QRCode, it seems not recognized, how can I get to make special characters get captured ?

Thanks!!

lvchaoin, 2021-08-24:
Finally changed to use logicalkey instead of codepoint, it seems the codepoint way can't support special keys.

library flutter_barcode_listener;

import 'dart:async';

import 'package:flutter/foundation.dart';
import 'package:flutter/material.dart';
import 'package:flutter/services.dart';

typedef BarcodeScannedCallback = void Function(String barcode);

/// This widget will listen for raw PHYSICAL keyboard events
/// even when other controls have primary focus.
/// It will buffer all characters coming in specifed `bufferDuration` time frame
/// that end with line feed character and call callback function with result.
/// Keep in mind this widget will listen for events even when not visible.
class BarcodeKeyboardListener extends StatefulWidget {
  final Widget child;
  final BarcodeScannedCallback _onBarcodeScanned;
  final Duration _bufferDuration;

  /// This widget will listen for raw PHYSICAL keyboard events
  /// even when other controls have primary focus.
  /// It will buffer all characters coming in specifed `bufferDuration` time frame
  /// that end with line feed character and call callback function with result.
  /// Keep in mind this widget will listen for events even when not visible.
  BarcodeKeyboardListener(
      {Key? key,

      /// Child widget to be displayed
      required this.child,

      /// Callback to be called when barcode is scanned
      required Function(String) onBarcodeScanned,

      /// Maximum time between two key events.
      /// If time between two key events is longer than this value
      /// previous keys will be ignored.
      Duration bufferDuration = hundredMs})
      : _onBarcodeScanned = onBarcodeScanned,
        _bufferDuration = bufferDuration,
        super(key: key);
  @override
  _BarcodeKeyboardListenerState createState() =>
      _BarcodeKeyboardListenerState(_onBarcodeScanned, _bufferDuration);
}

const Duration aSecond = Duration(seconds: 1);
const Duration hundredMs = Duration(milliseconds: 100);
const int lineFeed = 10;

class _BarcodeKeyboardListenerState extends State<BarcodeKeyboardListener> {
  List<int> _scannedCharCodes = [];
  DateTime? _lastScannedCharCodeTime;
  late StreamSubscription<int?> _keyboardSubscription;

  final BarcodeScannedCallback _onBarcodeScannedCallback;
  final Duration _bufferDuration;

  var str = "";

  final _controller = StreamController<int?>();
  _BarcodeKeyboardListenerState(
      this._onBarcodeScannedCallback, this._bufferDuration) {
    RawKeyboard.instance.addListener(_keyBoardCallback);
    _keyboardSubscription = _controller.stream
        .where((charCode) => charCode != null)
        .listen(onKeyEvent);
  }
[... 162 lines cut ...]
      return caps ? 'S' : 's';
    } else if (logicalKey == LogicalKeyboardKey.keyT) {
      return caps ? 'T' : 't';
    } else if (logicalKey == LogicalKeyboardKey.keyU) {
      return caps ? 'U' : 'u';
    } else if (logicalKey == LogicalKeyboardKey.keyV) {
      return caps ? 'V' : 'v';
    } else if (logicalKey == LogicalKeyboardKey.keyW) {
      return caps ? 'W' : 'w';
    } else if (logicalKey == LogicalKeyboardKey.keyX) {
      return caps ? 'X' : 'x';
    } else if (logicalKey == LogicalKeyboardKey.keyY) {
      return caps ? 'Y' : 'y';
    } else if (logicalKey == LogicalKeyboardKey.keyZ) {
      return caps ? 'Z' : 'z';
    } else {
      return "";
    }

    // case KeyEvent.KEYCODE_GRAVE:
    //     return caps ? '~' : '`';
    // case KeyEvent.KEYCODE_APOSTROPHE:
    //     return caps ? '"' : '\'';
  }
}

shaxxx, 2021-08-24:
Please use proper code formatting and code markdown tag, this is unreadable.