shaxxx/flutter_barcode_listener #6: Does not recognize codes, after the focus has been set to input field

lehnen, 2021-08-23:
Really nice idea to look at the input speed ;)

But to plugin only works fine for me until I use a text input that get's the focus and the on screen keyboard opens.
After that moment the barcode listener does not recognize the codes any more.

For reprodution I added
`Form(key: GlobalKey(), child: TextField(autofocus: false))` 
to the childs of the BarcodeKeyboardListener. 
Firstly the listener reacts on new codes. After the focus has been set to the input field, the barcode listener does not work any more.

KarolNiemiec, 2021-09-30:
same issue, after unfocus any input listener does not work any more

MFassmann, 2021-11-01:
@danielroek: I have the same issue as @lehnen, but also your workaround doesn't work.

lehnen, 2021-11-01:
I have success with my following implementation.
Just provide the callback and a focus node. In the calling Widget I perform a focusNode.requestFocus() every setState.

```
import 'dart:io';

import 'package:flutter/material.dart';
import 'package:flutter/services.dart';

typedef OnScanReceived = Function(String code);

class BarcodeKeyboardListener extends StatefulWidget {
  final OnScanReceived? onScanReceived;
  final FocusNode? focusNode;
  const BarcodeKeyboardListener({@required this.onScanReceived, @required this.focusNode, Key? key}):
        assert(onScanReceived != null),
        assert(focusNode != null),
        super(key: key);

  @override
  State<StatefulWidget> createState() => _BarcodeKeyboardListenerState();
}

class _BarcodeKeyboardListenerState extends State<BarcodeKeyboardListener> {
  final TextEditingController _controller = TextEditingController();

  @override
  void initState() {
    _controller.addListener(() {
      if (_controller.text.contains("\r") || _controller.text.contains("\n")) {
        String code = _controller.text.replaceAll("\n", "").replaceAll("\r", "");
        if (code.isNotEmpty) {
          widget.onScanReceived!(code);
        }
        _controller.clear();
      }
    });
    super.initState();
  }

  @override
  void dispose() {
    _controller.dispose();
    super.dispose();
  }

  @override
  Widget build(BuildContext context) {
    bool mobile = Platform.isAndroid || Platform.isIOS;
    return SizedBox(height: 1, child: mobile ? BarcodeMobileInput(focusNode: widget.focusNode!, controller: _controller) : BarcodeInput(focusNode: widget.focusNode!, controller: _controller));
  }
}

class BarcodeInput extends EditableText {

  BarcodeInput({
    @required TextEditingController? controller,
    @required FocusNode? focusNode,
    Key? key
  }):
    assert(controller != null),
    assert(focusNode != null),
    super(
      controller: controller!,
      focusNode: focusNode!,
      textAlign: TextAlign.center,
      style: const TextStyle(color: Colors.transparent, fontSize: 1),
      cursorColor: Colors.transparent,
      showCursor: false,
      backgroundCursorColor: Colors.transparent,
      autofocus: true,
      selectionColor: Colors.transparent,
      maxLines: 2,
      key: key
  );

}

class BarcodeMobileInput extends EditableText {

  BarcodeMobileInput({
    @required TextEditingController? controller,
    @required FocusNode? focusNode,
    Key? key
  }):
    assert(controller != null),
    assert(focusNode != null),
    super(
      controller: controller!,
      focusNode: focusNode!,
      textAlign: TextAlign.center,
      style: const TextStyle(color: Colors.transparent, fontSize: 1),
      cursorColor: Colors.transparent,
      showCursor: false,
      backgroundCursorColor: Colors.transparent,
      autofocus: true,
      selectionColor: Colors.transparent,
      maxLines: 2,
      key: key
  );

  @override
  EditableTextState createState() {
    return NoKeyboardEditableTextState();
  }
}

class NoKeyboardEditableTextState extends EditableTextState {
  @override
  void requestKeyboard() {
    super.requestKeyboard();
    SystemChannels.textInput.invokeMethod('TextInput.hide');
  }
}
```

MFassmann, 2021-11-01:
Thanks @lehnen, this works!

closed 2021-11-02 by danielroek