shaxxx/flutter_barcode_listener #7: Empty barcode

gunnerxd6, 2021-09-10:
I have a problem with this library, since although it manages to capture the barcodes, most of the attempts, the onBarcodeScanned function returns me an empty barcode, I have tried different EAN 13 barcodes but the result is inconsistent. I also tried to change the bufferDuration property but it doesn't seem to work either.
I'm working on a NQ1000-II android tablet with integrated barcode scanner.

shaxxx, 2021-09-11:
What do you mean by "result is inconsistent"?
Does it work at all? 
Does it work when you try to scan in some tex editor?
Does it work only sometimes?
Does it always work but result is not always correct?
What have you tried so far?
Please provide more details.

MFassmann, 2021-10-06:
I have a nearly similar problem. We are using a Honeywell CT60 with an internal scanner. This one we us with Wedge Method "Keyboard". The Problem is, that the digits aren't RawKeyUpEvent, which are needed. So just the "Enter" Button is fired an the returned barcode is empty.

solixone, 2021-10-12:
I set the scanner in Keyboard mode and it worked

shaxxx, 2021-10-19:
That was my idea. 
If I remember correctly there are some differences between handling KeyUp & KeyDown that need to be taken in consideration.
You can take a look at the commit history and see if something sticks out.

utpalruparel, 2021-11-24:
Any solution on this? I am getting empty barcode in Windows. 
On MacOS its working perfectly.

danielroek, 2021-11-24:
@utpalruparel 
I wasn't able to test this om Windows, as I run Fedora on my laptop. Didn't have the time to install a VM for this issue. 
However, I believe I found the source of the problem. 
This [line](https://github.com/shaxxx/flutter_barcode_listener/blob/daf7cee8191f378b80db5af11df589eeeb317119/lib/flutter_barcode_listener.dart#L119) 
```
        _controller.sink
            .add(((keyEvent.data) as RawKeyEventDataWindows).keyLabel);
```
should be changed to:
```
        _controller.sink.add(String.fromCharCode(
            ((keyEvent.data) as RawKeyEventDataWindows).characterCodePoint));
```

I've created a PR #9 could you please verify if this works for your implementation? You can do this by changing the dependency to:

```
  flutter_barcode_listener:
    git:
      url: https://github.com/danielroek/flutter_barcode_listener.git
      ref: fix/windows_charPoint
```

utpalruparel, 2021-11-24:
Thank you for the quick reply. I just tried with this dependency but not worked.

Actually now I am not getting blank barcode but getting blank spaces (same as string length of barcode)

parkmap, 2021-11-24:
worked on windows after change 

RawKeyUpEvent > RawKeyDownEvent

utpalruparel, 2021-11-25:
This solution is working for me too.  if (keyEvent is RawKeyDownEvent)

closed 2021-11-25 by gunnerxd6