shaxxx/flutter_barcode_listener #10: iOS scanning Problem

furkankurtulus, 2021-12-01:
When i trying on the ios 12 and 12.5 thats scanning correctly but when i trying on higher than 14 version thats scanning wrong.
Example: My barcode value is "102553"  but the output sometimes like "1022553","1025553".  They will giving  5 true data each 20-30 scanning. This problem not only for this package i got the that problem on the 'TextField' and 'TextFormField' widgets.

For the android, thats working absolutely correct, no error, no wrong scanning, problem only on ios14 and higher.

My Scanner: Opticon-2006
Note: I try with diffrent scanners but there is no solve.

shaxxx, 2021-12-01:
I did not understand completely.
Does barcode scanning works correctly when you're NOT using this plugin and scan while focused in text field?

wilhe897, 2021-12-04:
Have the same problem with IOS. The scanning works correctly outside the flutter app but inside it gives different values every time.

https://user-images.githubusercontent.com/94120467/144710691-f3759d83-345d-4d04-ad98-c8f4918bb4ca.mp4

mohammedebrahim1, 2021-12-06:
@furkankurtulus any updates I have the same problem ?

furkankurtulus, 2021-12-06:
I detected the problem, ,its about Flutter himself, Flutter fix this problem on higher versions but this versions not on the stable channel. Try change the flutter channel to beta and upgrade the flutter version, and problem will solved.

I'm using FLutter 2.6.0-12.0.pre-869 , you can use this version or higher.

closed 2021-12-15 by furkankurtulus