shaxxx/flutter_barcode_listener #11: Email string @ symbol is returned as "2"

blackbeario, 2021-12-23:
Hey, thanks for this package! I'm scanning 2D barcodes in a variety of formats, QR Codes and Data Matrix (doesn't seem to work in my tests with Aztec format but that's another issue).

I'm generating 2D barcodes with name, age, and email strings. While on the Flutter stable channel, the results were returned wrong nearly every time. But after switching to the Flutter Beta channel the results are correct except for the email string. Specifically the @ symbol is returned as "2".

If I add a barcode with the email me@gmail.com, the plugin returns me2gmail.com every time. I don't know much about this plugin yet, but will see if I can figure it out.

Flutter (Channel beta, 2.9.0-0.1.pre, on macOS 11.2 20D64 darwin-x64, locale en-US)

danielroek, 2022-01-26:
Hey @blackbeario, 

In the plugin this is the line that handles the MacOS keyEvent:
```dart
...
      } else if (keyEvent.data is RawKeyEventDataMacOs) {
        _controller.sink
            .add(((keyEvent.data) as RawKeyEventDataMacOs).keyLabel);
      }
```
The problem here is that we listen to the `keyLabel`, which is '2'. I've created a PR([#13](https://github.com/shaxxx/flutter_barcode_listener/pull/13)) that listens to `character` instead. I don't have access to a MacOs device any time soon, would you be able to test if this works?

You'd be able to test this by changing the dependency in pubspec.yaml to:
```
  flutter_barcode_listener:
    git:
      url: https://github.com/danielroek/flutter_barcode_listener.git
      ref: mac_os_characters
```

shaxxx, 2022-11-11:
This is merged. Closing.

closed 2022-11-11 by shaxxx