shaxxx/flutter_barcode_listener #12: Get empty after scan in windows

TungWang, 2022-01-20:
Hi!
Thank you for create this plugin.
I got an empty string after scan barcode.
Is there any way to solve this problem?

VAbbas, 2022-01-22:
i too got empty string after scan in windows

danielroek, 2022-01-26:
It seems that in some cases Windows uses `RawKeyDownEvent` instead of the implementations `RawKeyUpEvent`. I've created a PR, #14 , in which it is possible to set the use of `RawKeyDownEvent` by setting the `useKeyDownEvent` to true. Keep in mind that using this on non-Windows might result in issues. A possible way to overcome this would be to use `Platform.isWindows` if you are supporting multiple operating systems. 

Would you be able to see if the PR I created works as intended? You can test this by changing the dependency in pubspec.yaml to:
```
  flutter_barcode_listener:
    git:
      url: https://github.com/danielroek/flutter_barcode_listener.git
      ref: feature/key_down_event
```

tosyak, 2022-03-03:
Hi!
Want to thank you for your package!
I also got this problem in windows and i've tried to use your dependency from git, but they also did not work.
So i took your package and replaced method and used it localy. This way that worked!

danielroek, 2022-03-03:
Hey @tosyak , 

Could you provide the method you changed to make it work? That way we'd be able to improve this package.

shaxxx, 2022-03-03:
I'm working on this at this moment

shaxxx, 2022-03-03:
Hopefully fixed in 0.1.3
Please reopen issue if you're still having problems.

closed 2022-03-03 by shaxxx