shaxxx/flutter_barcode_listener #15: Remove green border

berrybana, 2022-02-21:
Hi, Thanks for this library.
In my case,  green border appears after scanning the barcode.
Is there any way to remove this?

berrybana, 2022-02-21:
Attach screen shot
[image]

danielroek, 2022-02-22:
Hi @berrybana, I'm not quite sure what green border you mean. Would you be able to provide us with some code to reproduce the issue and your `flutter doctor` output?

berrybana, 2022-02-23:
Green border like screenshot below.
I think green border is focus highlight. 
So I added this code to style.xml
<item name="android:defaultFocusHighlightEnabled">false</item>
then, border disappeared!!

[image]