shaxxx/flutter_barcode_listener #17: I can't see the value of barcode on toast When I scan barcode when keyboard on Korean..

SooHoon95, 2022-02-23:
I Want to see the values of barcode immediately on toast message.
But when I use this pack during setting keyboard for Korean,
the value doesn't have korean charaters

ex) value of barcode : 100Barcode-123123

when I scan in Eng -> 100Barcode-123123
but in Korean -> 100-123123

so I inspected the package.

I expect that when I use in iOS , the values return keylabel types.
and keyLabel is defined in doc

'it will be missing special processing for Unicode strings for combining characters and other special characters, and the effects of modifiers.'.

so I don't know what i have to do next.
I need your help guys

danielroek, 2022-03-03:
Hi @SooHoon95 , 

When using `keyLabel` we found that some characters where displayed wrong, for example a '2' instead of '@'. Because it ignores modifiers. This is why we chose to use `character` instead. I'm also not sure if using `keyLabel` would solve the issue. You could test this by using version [0.1.1](https://github.com/shaxxx/flutter_barcode_listener/blob/master/CHANGELOG.md#011---web-and-desktop-support), this version did use the MacOS `keyLabel`.

SooHoon95, 2022-03-10:
I already use version 0.1.1.
Can it be wrong about Unicode or surport problem language of keyLabel??
really bother me..

shaxxx, 2022-11-11:
There's a limit to first 255 characters in ASCII table since working with unicode need extra care and testing on different platforms.

` if (keyEvent.logicalKey.keyId > 255 &&
        keyEvent.data.logicalKey != LogicalKeyboardKey.enter) return;`

Maybe you can try without this?