shaxxx/flutter_barcode_listener #21: Windows Desktop can not get full code

cwangfr, 2022-04-15:
I have a strange problem, in some Windows systems, the complete barcode cannot always be scanned, and the results in release mode and debug mode are different.
My code is this
`BarcodeKeyboardListener(
              bufferDuration: const Duration(milliseconds: 500),
              useKeyDownEvent: true,
              onBarcodeScanned: (barcode) {
                debugPrint(barcode);
                if (barcode.length == 12) {
                  barcode = "0" + barcode;
                }
                controller.barcode(barcode: barcode);
              },
              child: barcodeList(),
            )
`
i have 3 barcode 

0755441231925
20147426
3700408421526

in debug mode result 
[image]

in release mode result 
[image]

utpalruparel, 2022-05-07:
Hello.. have you found any solution for this?

deanbot, 2022-05-14:
I'm assuming this is a flutter issue and should be raised with flutter. The logs I'm seeing indicate that flutter is throwing out chars if an identical char is entered too quickly, i.e. if scanning 10005: 

```sh
I/flutter ( 4370): char event: 1
I/flutter ( 4370): char event: 0
I/chatty  ( 4370): uid=10136(com.example.myapp) 1.ui identical 1 line
I/flutter ( 4370): char event: 0
I/flutter ( 4370): char event: 5
```

deanbot, 2022-05-14:
Note, if you are able to configure scanning inter character delay this workaround is working for me https://github.com/flutter/flutter/issues/79849#issuecomment-1007147194

shaxxx, 2022-11-11:
Closing this since it's issue with Flutter, not this plugin.

closed 2022-11-11 by shaxxx