shaxxx/flutter_barcode_listener #24: Issue with Keyence reading

selmaskopljakovic, 2022-06-24:
Hello, 

I tried to use this package for the Keyence device. My app runs at Keyence, but I can't detect scan triggers to use data with your example. 
Your idea is exactly what I need, scanning without focusing on TextBox and without a mobile keyboard, just to listen to triggers from the device. 
Do you have any idea why this is not working?
Thank you!

BIG-BEARC, 2022-06-30:
Hello,
This was my solution,hope it works for you
[https://github.com/shaxxx/flutter_barcode_listener/pull/25#issue-1289656467](url)

shaxxx, 2022-06-30:
Solultion from @Bisc0tt1 is the right one. Check how you can configure scanner to emit linefeed end as it is industry standard.

selmaskopljakovic, 2022-06-30:
Thank you guys so much!
I will try it and come back with a feedback

selmaskopljakovic, 2022-07-05:
Hello, ok, this is how I solved the reading barcode with Keyence:
1. Settings of the device - Turn on Keystroke
2. Settings of the device - Set terminator to Enter Key
3. This is how I read data 
`(keyEvent.data) as RawKeyEventDataAndroid).keyLabel`

Have duplication, like 22334455 instead 2345. But That's easy to solve.

The only issue I'm struggling with is turning off the keyboard. I tried all that could be found on google, but still no good solution.

shaxxx, 2022-11-11:
Closed as solved.

closed 2022-11-11 by shaxxx