shaxxx/flutter_barcode_listener #28: Caps Lock error on Android 6

Wwwolfgang, 2022-07-04:
First, very awesome library.
Im using it on a Android device with an external Barcode scanner. 
Every time I scan something, im getting this error 
`Exception caught by services library 
══════════════════════════════════
Attempted to send a key down event when no keys are in keysPressed. This state can occur if the key event being sent doesn't properly set its modifier flags. This was the event: RawKeyDownEvent#82bce(logicalKey: LogicalKeyboardKey#00104(keyId: "0x100000104", keyLabel: "Caps Lock", debugName: "Caps Lock"), physicalKey: PhysicalKeyboardKey#70039(usbHidUsage: "0x00070039", debugName: "Caps Lock"), repeat: false) and its data: RawKeyEventDataAndroid#d1f09(flags: 8, codePoint: 0, plainCodePoint: 0, keyCode: 115, scanCode: 58, metaState: 0)
'package:flutter/src/services/raw_keyboard.dart':
package:flutter/…/services/raw_keyboard.dart:1
Failed assertion: line 688 pos 7: 'event is! RawKeyDownEvent || _keysPressed.isNotEmpty'`

Any idea what the problem could be? Is it a problem of the library or off the scanner?
Im working on Flutter version 3.0.3

danielroek, 2022-10-03:
This seems to be an issue with Flutter itself. I've seen some of these myself when running linux applications in debug-mode, even without this library.

shaxxx, 2022-11-11:
This has gone stale. Closing it.

closed 2022-11-11 by shaxxx