shaxxx/flutter_barcode_listener #29: Q:

craigbrown-nist, 2022-07-18:
Is it possible to have the widget gain focus, or to defocus any other widget(s) once the barcode has been determined? 
 
I have a build that has some input widgets above a visibility widgets and the barcode scanner as used in the example and it works great (windows version) if you click off the other widgets that take input. 

However, if focus is in one of those other input fields the also get the barcode as keypresses.

I have tried adding in a 'WidgetsBinding.instance.focusManager.primaryFocus?.unfocus();' call just above the barcode scanner, but it doesn't work.

Any one have a suggestion in how to do this?

danielroek, 2022-10-03:
Not really sure what you are trying to achieve without some code examples

shaxxx, 2022-11-11:
Closed due to missing information.

closed 2022-11-11 by shaxxx