shaxxx/flutter_barcode_listener #31: only reading Lowercase on android.

baconjuggling, 2022-10-03:
Scanning QR codes on android is returning the barcodes value, but only  in lowercase, I've had a little play but cant seem to find a fix, any ideas?

danielroek, 2022-10-03:
I think the scanner probably send the small letter with a `shift-key modifier`. This is what makes this library a bit hard to generalize, every scanner is different and we can't really support anything else than UTF-8.

baconjuggling, 2022-10-16:
I made a basic fix for my project, that could potentially be implemented, perhaps with a optional caseSenstive: bool parameter.

BangPei, 2022-10-18:
@baconjuggling hi, can u share me the code?
i've tried to uppercase the character but doesnt work.
thank you

baconjuggling, 2022-10-18:
> @baconjuggling hi, can u share me the code?
> i've tried to uppercase the character but doesnt work.
> thank you

It's available on my fork.
it's currently only a fix for android though.

BangPei, 2022-10-18:
> > @baconjuggling hi, can u share me the code?
> > i've tried to uppercase the character but doesnt work.
> > thank you
> 
> It's available on my fork. it's currently only a fix for android though.

Thanks, i will try it.

shaxxx, 2022-11-11:
Closed due to lack of feedback.

closed 2022-11-11 by shaxxx