shaxxx/flutter_barcode_listener #38: Need device information

lucreck, 2023-04-05:
I usd multi barcode scanners, so I want to know barcode scanned by what device.
So how I get the device info.
Thank you.

SeanKD, 2023-06-15:
on the zebra (MC3300) scanner I have onBarcodeScanned method does not trigger at first but if i close the app and reopen it it works fine, did you come up with a fix for this?

fgarcia78, 2023-07-28:
We're having a similar problem: the components scans just one code and it gets corrupted (the widget resets any customization, like background color); the only way to make it work again is reopening the app.

I want to debug the app from a real device (Android tablet) to try to identify the problem that is happening, but I haven't had the time yet.

SeanKD, 2023-09-20:
@fgarcia78 any luck with this?

shaxxx, 2023-11-08:
No feedback.

closed 2023-11-08 by shaxxx

rymesaint, 2024-01-15:
is there any update on this?