shaxxx/flutter_barcode_listener #41: Scanned barcodes and qr codes is in lowercase

davepot04, 2023-06-07:
[image]

values of scanned barcodes and qr codes are all in lower case but when I don't use the package and only used a TextFormField the value is correct, i think it can't detect the if the keyevent has a shift key pressed with other character

(sorry for my bad english 😅)

fuadreza, 2023-07-13:
Hi @davepot04 , 

I made an update for the scanned character to be case sensitive You can see on my forked version here:

https://github.com/fuadreza/flutter_barcode_listener

SchrodingersGat, 2023-10-24:
> Hi @davepot04 ,
> 
> I made an update for the scanned character to be case sensitive You can see on my forked version here:
> 
> https://github.com/fuadreza/flutter_barcode_listener

@fuadreza it works really well - any chance you will be pushing to https://pub.dev/ ?

fuadreza, 2023-10-24:
I have already open PR #44 , but it seems this package maintainer haven't responded yet.

SchrodingersGat, 2023-10-25:
Yeah, no updates since March last year, might be dead?

shaxxx, 2023-10-25:
Well, it's not so simple as you might think.
Plugin is used by many people and if current working condition changes it's a breaking change, and people hate breaking changes.
I'm simply waiting for feedback, and as things are, not many people are requesting this.
Beside, you don't need pub.dev, you can always specify direct git link in your pubspec.yaml.

With all this said, just fix PR and I'll merge it.

SchrodingersGat, 2023-10-25:
@shaxxx all fair points, hope I didn't come across poorly :)

fuadreza, 2023-10-25:
> Well, it's not so simple as you might think. Plugin is used by many people and if current working condition changes it's a breaking change, and people hate breaking changes. I'm simply waiting for [...]

Oh I see, I will add some option to make it optional and won't break changes.

shaxxx, 2023-11-08:
Added in 0.1.4

closed 2023-11-08 by shaxxx

PraveenGangula1392, 2024-06-13:
For me the issue is vice-versa , on scan ; is getting converted to : 
Example S24-00318;A

Output   S24-00318:A