shaxxx/flutter_barcode_listener #56: After i scan its opening the keyboard

patelnirav48, 2024-05-26:
1. Scan the barcode
2. Its scans but opening keyboard

Platform - Android

andresthinkme, 2024-05-27:
I am having the same issue.

patelnirav48, 2024-05-28:
@shaxxx Could you please help here? 
Keyboard get open automatically, after i scan the barcode.

PraveenGangula1392, 2024-06-12:
@shaxxx 
When we do not put focus on any text field and scan everything works perfectly. But let's say i have a text field in my form i entered some data and then scan a barcode then the keyboard shows up, even after i tap outside the text field and scan still the keyboard shows up..

seba-i, 2025-01-04:
You need to muck about with the device keyboard settings - on Android it was something like "don't open virtual keyboard when physical keyboard...". Barcode scanners operate as a physical keyboard.